riscv/sail-riscv #668: Define operators for physaddr and virtaddr

Timmmm, 2025-01-06:
The new `physaddr` and `virtaddr` types are quite nice, but they do make some code uglier. For example instead of

```
handle_mem_exception(vaddr - offset), e);
```

we have to do

```
handle_mem_exception(virtaddr(virtaddr_bits(vaddr) - offset), e);
```

I think we can overload + and - (and maybe others) so that they work with `physaddr` and `virtaddr` too.

trdthg, 2025-01-07:
I tried this before, but due to the order of file dependencies, I felt the effect was not very ideal

The current compilation order is `PRELUDE = prelude.sail ... prelude_mem.sail`

- General overloads are defined in `prelude.sail`
- If you want to overload operator for `physaddr` and `virtaddr`, you must define it in `prelude_mem.sail` or after it, possibly in a new `prelude_operator.sail`. But this might make us move all the operator overloads from `prelude.sail` to `prelude_operator.sail`, which looks very messy.

And will sail support things like

- `overload operator xxx += {  }`
- `overload operator xxx for type`

Timmmm, 2025-01-07:
Yeah it's described [here](https://alasdair.github.io/manual.html#_ad_hoc_overloading). I think adding them just after the `newtypes` is fine.

PR #672 (2025-01-14): Overload operators for physaddr and virtaddr
try resolve #668 

Added operator overloading for `physaddr` and `virtaddr`

I think we don't need to implement `==` and `!=`? 
Because `eq_anything` (I just discovered this) has already been introduced in `prelude.sail`.

https://github.com/riscv/sail-riscv/blob/c0ebea2ff14e6a39d53161c3e51dcbabe3e9a0f1/model/prelude.sail#L17
closed 2025-01-14 by Timmmm