riscv/sail-riscv #727: Remove `ENABLE_SPIKE`

trdthg, 2025-02-11:
https://github.com/riscv/sail-riscv/pull/725#discussion_r1948687529

The `ENABLE_SPIKE` parts are quite outdated, I'm removing them.

But I don't know what spike is used for. Does anyone still use it?

Timmmm, 2025-02-11:
Looks like it was once used for doing comparisons against Spike... but I've certainly never tried it, and it isn't tested in CI so it's almost certainly broken. I think in future we will do this via a proper library interface anyway so yeah I think we should remove it.

pmundkur, 2025-02-11:
Yeah, this is pretty old and bitrotted.  There are comments in the Makefile about it which can also be removed.

jordancarlin, 2025-02-12:
Looks like there is a chunk about it in the README that should be stripped out too.

PR #732 (2025-02-13): Remove support for linking C emulator to Spike
resolve #727 

- remove all `#ifdef ENABLE_SPIKE`
- remote `init_spike()`, `init_check()`, `compare_states()`
- remove things about linked to Spike in README and Makefile
closed 2025-02-13 by Timmmm