riscv/sail-riscv #737: Make executable name generated by CMake backward compatible with the old Makefile

Barabas5532, 2025-02-13:
The new cmake build system #551 #647 generates an executable named riscv_sim_rv32d. Could you make it generate the old name so it is easier for users to upgrade?

It could be a cmake option that installs a symlink with the old name so that people already updated don't get broken again.

There are [many projects and tutorials](https://github.com/search?q=riscv_sim_RV32&type=code) using the old name already.

Timmmm, 2025-02-13:
Hmm most of those use `c_emulator/riscv_sim_RVxx` though, which wouldn't work anyway since the executables are generated in a build directory now.

The other wrinkle is that the name is going to (hopefully) change *again* when we build a single emulator that can do all xlen & flen combinations.

I kind of think enabling a CMake option is just as much work as editing the name... I guess we could put a symlink in the repo from `c_emulator/riscv_sim_RV32` to `../build/c_emulator/riscv_sim_rv32d`? Might be confusing that it starts off broken though...

Barabas5532, 2025-02-13:
I don't mind the change in the location of the built executable from c_emulator to the build folder with cmake. The location of the executable can be updated in the build script when switching to cmake.

Actually I used cmake to install, which abstracts the built executable location completely: https://github.com/Barabas5532/crush/commit/9888719ff19609a6383f583267f3145acc9f06b1#diff-f34da55ca08f1a30591d8b0b3e885bcc678537b2a9a4aadea4f190806b374ddcL38. If I didn't do this, I could have updated the `mv` command to rename the executable for compatibility as well. 

I wouldn't hardcode the symlink to build. Users can specify any build folder name for cmake. It could be fine if the build guide explains to use exactly "build" as the folder name.