riscv/sail-riscv #742: Track Cryptography Extensions

trdthg, 2025-02-17:
Hi, I would like to discuss about developing Cryptography Extensions

Cryptography Extensions includes 15 sub extensions, current development status:

| Extension | Status | PR/Issue |
| ----------| ----- | -----|
| Zvbb |  |  |
| Zvbc |  |  |
| Zvkb | PR in Review | #236 |
| Zvkg | PR in Review | #241 |
| Zvkned | PR in Review | #234 |
| Zvknh(ab) | PR in Review | #235 |
| Zvksed | PR in Review | #235 |
| Zvksh | PR in Review | #248 |
| Zvkn |  |  |
| Zvknc |  |  |
| Zvkng |  |  |
| Zvks |  |  |
| Zvksc |  |  |
| Zvksg |  |  |
| Zvkt |  |  |

The extensions in review above were developed by @charmitro (thanks for your great work), and currently they need some rebase and recheck against the spec.

We (currently it is mainly @trdthg, @KotorinMinami, @rez5427 and some others who are still learning) will continue to work in sail model for a long time. (we are also continuously doing some sail teaching and recruiting) So we are interested in develop and maintain the entire Cryptography Extensions

jordancarlin, 2025-02-17:
Looks like this is specifically referring to the vector crypto extensions (all of the scalar crypto extensions are merged I believe).

Many of the current PRs seem to be targeting the older vector-dev branch, so likely need some substantial rebasing. It also looks like they were developed based on draft versions of the spec, so as you mentioned they definitely need to be checked against the ratified version for any changes. 

Most of these are optional extensions for RVA23, so getting them into the Sail model is definitely a priority. It probably makes sense to track them over on the [Extension Roadmap Wiki Page](https://github.com/riscv/sail-riscv/wiki/Extension-Roadmap).

KotorinMinami, 2025-02-17:
Confirmed, all scalar extensions are supported.

We can start with the previous review's extensions, submitting each as a PR and keeping maintaining until they are merged.

jordancarlin, 2025-02-23:
I've transferred all of this to the [Extension Roadmap Wiki](https://github.com/riscv/sail-riscv/wiki/Extension-Roadmap#vector-crypto-extensions) to make it easier to track with all of the other extensions.

closed 2025-02-25 by trdthg