riscv/sail-riscv #745: Test build_emulators.sh and DOWNLOAD_GMP in CI

Timmmm, 2025-02-17:
We have a couple of features that aren't tested in CI and could easily bitrot:

* `-DDOWNLOAD_GMP` to download GMP and compile it from source.
* The `./build_emulators.sh` script.

The latter uses the former, so we can just add `./build_emulators.sh` to CI.

jordancarlin, 2025-03-31:
Took a look at this, and apparently gmplib.org blocks GitHub Actions IP addresses, so this can't currently be tested in CI: https://github.com/actions/runner-images/issues/7901. Seems like one solution is to download from https://ftp.gnu.org/gnu/gmp/ instead.

Arielfoever, 2025-04-22:
Suggest doing these:

- add a ci to build everything with ./build_emulators.sh
- add some mirrors of GMP

I wiil do the first one now