riscv/sail-riscv #756: Update README for new CMake build system

Alasdair, 2025-02-26:
I think the README needs to be updated for the new CMake build system as says:
```
make -C build riscv_sim_rv64f_rvfi
```
but this doesn't work until you've run CMake at least to get the build directory. I think the intent behind the current wording is to run the `build_simulators.sh` script first, but I don't think this is obvious enough.

Timmmm, 2025-02-26:
Hmm I couldn't think of a good way to word this so what if we just change the code block to this?

```
$ ./build_simulators.sh
$ make -C build riscv_sim_rv64f_rvfi
```

jordancarlin, 2025-02-26:
What about having the first line directly invoke cmake to avoid unnecessarily building the standard simulators first.

Timmmm, 2025-02-26:
That gets a bit complex for a first command the user ever runs maybe...

```
mkdir -p build
cmake -S . -B build -DCMAKE_BUILD_TYPE=RelWithDebInfo -DDOWNLOAD_GMP=TRUE
make -C build riscv_sim_rv64f_rvfi
```

jordancarlin, 2025-02-26:
The `mkdir`command shouldn’t be needed (cmake creates that directory automatically).

> That gets a bit complex for a first command the user ever runs maybe...

I assume most users won’t be using the rvfi version, so not sure that this is a huge concern. Also, once we have the pre-compiled releases I think we should change this section to direct users to those and have a separate developers (or something along those lines) section that explains how to actually build the model. Most people should not need to build from source themselves.

Timmmm, 2025-02-26:
So this?

```
cmake -S . -B build -DCMAKE_BUILD_TYPE=RelWithDebInfo -DDOWNLOAD_GMP=TRUE
make -C build riscv_sim_rv64f_rvfi
```

I'm ok with that if you guys think it is better.

jordancarlin, 2025-02-26:
> So this?
> 
> ```
> cmake -S . -B build -DCMAKE_BUILD_TYPE=RelWithDebInfo -DDOWNLOAD_GMP=TRUE
> make -C build riscv_sim_rv64f_rvfi
> ```

That’s what I was thinking. I think we should definitely go with one of the options discussed in this thread and this one would be my preference to avoid redundant/unnecessary compilation, but I’d be fine with either.

PR #893 (2025-04-29): Update README build instructions and general cleanup
Fixes #756