riscv/sail-riscv #775: Make reserved behaviour configurable

Timmmm, 2025-03-07:
Currently the Sail model is mostly written so that if code triggers some reserved behaviour the model will do something sensible, e.g. an illegal instruction exception, ignoring CSR writes, etc.

An alternative would be to immediately abort execution. However I think there are use cases where we don't want to do that. Alasdair mentioned that some code relies on reserved behaviour (e.g. bootloaders), and I would be *really* surprised if you could e.g. boot Linux without hitting any.

I think there are use cases for both modes, so I think we want a global flag with these two options:

1. Reserved behaviour will immediately abort the model.
2. The model will take some realistic action that a sensible real CPU would do.

I don't think there's an exhaustive list of reserved behaviours, but based on a rough scan of the ISA manual:

1. `mtvec.MODE` other than Direct (0) / Vectored (1), at least until we support CLIC.
2. `pmpcfg` with R=0, W=1
3. `miselect` with the most-significant bit clear (not implemented in the model yet)
4. Writing reserved values to any WLRL field.
5. Executing non-existent instructions.
6. Accessing non-existent CSRs.
7. Odd-numbered registers in `amocas` `rs2` and `rd`
8. `xenvcfg.CBIE=0b10`
9. Reserved `frm` modes: "The behavior of floating-point instructions that depend on rounding mode when executed with a reserved rounding mode is reserved, including both static reserved rounding modes (101-110) and dynamic reserved rounding modes (101-111)."
10. `rm` other than `RTZ` for `fcvt.w.d`
11. Odd-numbered registers for RV32Zdinx
12. `rlist` 0-3 for `cm.push/pop`
13. `jvt.mode` other than 0
14. `vsew[2] == 0b1`
15. "The use of vtype encodings with LMUL < SEWMIN/ELEN is reserved"
16. Loads more stuff in vector

Some of these are really the general case of writing reserved values to any WARL, which is sort of slightly specified (the implementation can't raise an exception, and has to deterministically return a legal value).

The language in the spec is often a big woolly around "reserved", e.g. the `fence` instruction has a load of reserved encodings but the spec also tells you what implementations should do for those cases.

radimkrcmar, 2025-03-07:
Aborting is sometimes an acceptable implementation, but the ISA doesn't allow it in all cases (e.g. WLRL).
Maybe a emitting warning as a micro-architectural side-effect would be more appropriate behavior, because that wouldn't make the sail implementation incorrect.  As a side note, software is expected to rely on some implementation-specific behavior (e.g. WARL), so we should also be able to suppress some warnings.

There are also official extensions to specify the behavior (e.g. Ssstrict to trap on reserved instructions/CSRs).
I think it would be good to have similar finer-grained configuration options for each implementation-specific behavior.

Sail model could use the unified database, because they did a lot of work to describe the implementation-specific behavior.
Random example: https://github.com/riscv-software-src/riscv-unified-db/blob/main/arch/ext/S.yaml

allenjbaum, 2025-03-08:
There are at least 3 different spreadsheets of architectural options, which
are subsets of implementation defined behaviors.
( from Synopsys, ACTs, Harvey Mudd, and Qualcomm). We do need to merge
those
Pedantic hat on:  You can say nothing about what an implementation specific
behavior is; it could do absolutely anything:
   trap alias to something else, hang, halt and catch fire, etc.
   Architectural options are where the implementation can select between
defined behaviors (e.g. trapping or not)
  Pedentic Hat Off
One of the problems with that is that it is left unstated whether the
selection is fixed, or deterministic (depending on some architectural state)
  or nondeterministic. I am thinking of trying to get a fastrack extension
(similar to Ssstrict) that requires all unnamed architectural options in
the spec to be fixed unless explicitly   defined otherwise (e.g. vector
tail agnostic, vector unordered indexed store)

Timmmm, 2025-03-10:
Discussed in meeting. We're going to wait for the new config system and then add separate flags for each case we encounter.

ibvqeibob, 2025-12-15:
Add miselect CSR (0x350) and handle MSB=0 reserved values #1424

allenjbaum, 2025-12-26:
IF not otherwise defined, the default mapping of illegal->legal value in a CSR field field should be to ignore the write (leave the current legal value unchanged).
Just defining what the legal and illegal values are will be hard enough. 

riscv-config defined 8 different "standard" methods it expected (and we expected Sail to eventually handle:

unchanged: The value remains unchanged to the current legal value.
value	A fixed (legal) value can also be specified
next higher/lower legal value	(or hiigst/lowest if it is already the highest/lowest
closest legal value, going higher/lower in case of a tie:
highest/lowest legal value

There was one other special case, but it turns out it wasn't needed.

Note that a WARL field will often by split into multiple subfields, e.g. CSRs holding addresses may have the lower bits by fully RW, and the upper bits be RdOnly0 or RdOnly1s

allenjbaum, 2025-12-26:
> Why? I suspect we won't have a default anyway - you'll have to explicitly
> choose.
> What's the motivation for this? Seems a bit odd to me, especially for
> "enum" style fields.

In theory, they'd choose, but if they implement this default behavior
(which is common, though I'd prefer if it were mandated), then they don't
have to specify the behavior for every single WARL field.

SAIL must be told what is legal for a WARL field - I think that is required
, and it's hard enough - but we wanted to constrain the mapping from
illegal->legal
At the time we defined this, we were trying to come up with the set of
actions that would cover 99.9% of what we could conceive anyone doing.
That "next" pair, and the "nearest_ties lower/high are outliers, I'm
reasonably sure.

In theory, of course, it doesn't matter, because anyone writing SW that
affects a WARL field should/will need to read it back to ensure that it got
written properly,
and if it wasn't, what they should do about it -- but writing tests where
the answer could be different than what SAIL does is problematic, so the
idea is to put burden on Sail once
(e.g. a single common, parameterized legalization routine), rather than
putting the burden on every test from now one.