riscv/sail-riscv #781: Maybe revert regidx newtypes?

Timmmm, 2025-03-10:
So, I benchmarked compilation before/after #617 and the results are pretty bad at least on my laptop. The size of the C file is also dramatically increased:

| | Before | After |
|-|-|-|
|Time| 37s | 8m2s |
| C size | 12 MB | 42 MB |

This is with Clang 18.1.8. The extra time is in C compilation.

It's not quite as bad in CI but it's still more than doubled: [6m31s before](https://github.com/riscv/sail-riscv/actions/runs/13764245527), [13m31s after](https://github.com/riscv/sail-riscv/actions/runs/13764550192).

I was a bit on the fence about it anyway since it's so verbose. It does improve type safety a bit and makes E support more "correct"... but I don't think it's worth it given this. E is a subset of non-E anyway, so we technically "support" it already (though maybe not in a useful way for some use cases, e.g. SV formal equivalence checking).

I haven't benchmarked runtime performance but given the huge amount of extra code I'd be slightly surprised if Clang/GCC can completely eliminate it all.

Maybe the newtype codegen can be improved @Alasdair ? Or should we just do it without newtypes, so we still have `regidx`, `fregidx`, `vregidx` but they're not newtypes, i.e. `type fregidx = bits(5)`?

@jordancarlin @arichardson @nwf what do you think?

arichardson, 2025-03-10:
I like this change and we should keep it. IMO this should be fixed in the sail codegen. But considering the impact here, maybe we need to revert it until there is a sail release with the fix.

jordancarlin, 2025-03-10:
@Alasdair mentioned that it might be possible to optimize out most of the extra wrappers that are added for these conversions (https://github.com/riscv/sail-riscv/pull/762#issuecomment-2711232567). If that can be done in time for the upcoming Sail 0.19 release then I'd say let's keep it. Otherwise it might make sense to revert because that is a huge increase in compilation time.

arichardson, 2025-03-10:
Perfect, in that case I'd say we use the upcoming 0.19 release ASAP and keep the change. reverting will cause a lot of churn for existing PRs so I'd say we deal with the excessive compile time for a few days?

Timmmm, 2025-03-10:
Yep sounds good to me. Thanks @Alasdair. Thalasdair.

closed 2025-03-10 by Timmmm

Timmmm, 2025-03-11:
I think there is some still some regression... but it's definitely massively improved.

I now get 1m4s vs 1m34s (don't ask my why it isn't 37s any more; tbf I'm benchmarking on a crap laptop).

Also the code output is still a little bigger -  286k lines vs 334k. Maybe we'll just have to live with that.

Alasdair, 2025-03-11:
There might be some additional trivial copies from the type to itself, rather than wrap/unwrap calls for the constructor.

The PR I had also fixed the too-deep nesting issue, and that might affect how the C++ compiler handles the code.

jordancarlin, 2025-03-11:
@Timmmm @Alasdair, looks like CI is still quite a bit slower. 6:31 before any of the changes, 13:31 with the new regidx and Sail 0.18, and 9:58 with the new regidx and Sail 0.19.

Alasdair, 2025-03-12:
Do you know where the slowdown is? I think the regidx change shouldn't really be making the C++ compiler work that much harder. If it is we might need to implement a better copy elision pass in Sail, but that might just be shifting the performance cost around rather than actually fixing anything.

It's been a while since I did any performance profiling of Sail itself, so it could have just gotten a bit slower with some of the refactorings and changes I did while working on Sail to SystemVerilog.