riscv/sail-riscv #795: Help with build issue

ved-rivos, 2025-03-15:
I am facing this build issue. Please help:

```
ved:$ which sail
/home/ved/.opam/5.1.0/bin/sail
ved:$ sail -v
Sail 0.17.1 (sail @ opam-v2.1.5)
ved:$ opam --version
2.1.5
```

When I run build_simulators.sh I get the following error:
```
ved:$ eval $(opam config env)
ved:$ env DOWNLOAD_GMP=FALSE ./build_simulators.sh
-- Found sail: /home/ved/.opam/4.08.0/bin/sail
CMake Error at sail_runtime/CMakeLists.txt:1 (execute_process):
  execute_process error getting child return code: No such file or directory

-- Configuring incomplete, errors occurred!
```

Sail is at /home/ved/.opam/5.1.0/bin/sail but the build script think its at /home/ved/.opam/4.08.0/bin/sail

jordancarlin, 2025-03-15:
There have been several recent updates to the model, and we now require Sail version 0.19. Unfortunately, the warning message indicating the sail compiler version is too old was only introduced in Sail 0.18.

Assuming you are on a Linux distro, I recommend transitioning away from opam to the precompiled binary versions of Sail. The most recent one can be found at https://github.com/rems-project/sail/releases/tag/0.19-linux-binary.

Alasdair, 2025-03-15:
Yes, you'll need to update your sail version.

However, I think cmake seems to remember the last installed location, so you may need to re-run the configure step or just nuke the build directory so it figures out where the updated version is. I ran into a similar issue when I was trying the new release.

ved-rivos, 2025-03-15:
Where can I find instructions to install the tarball

Alasdair, 2025-03-15:
You should just be able to extract the tarball, then add the included `bin` directory to your `$PATH`.

ved-rivos, 2025-03-16:
Thanks much for your help. I had to build cmake from source as well as the project now requires >= 3.20 and the Ubuntu package is 3.16.

closed 2025-03-16 by ved-rivos

Timmmm, 2025-03-16:
Weird trick but there's a CMake package in pip that is kept up to date. `python3 -m pip install cmake`. Though on recent distros it will shout at you and you have to use `--break-system-packages` or something like that.

I haven't tried it but I believe uv can install Python tools without angering the Debian gods, so this might work as a quick & reliable way to get a recent CMake.

```
curl -LsSf https://astral.sh/uv/install.sh | sh
uv tool install cmake
```

jordancarlin, 2025-03-16:
You can also download the precompiled cmake binaries from https://github.com/Kitware/CMake/releases/download/v3.31.5/cmake-3.31.5-linux-x86_64.tar.gz to avoid dealing with pip or uv. Just extract it and add it to your path. I’ve used this version on Ubuntu 20.04 with no issues.

arichardson, 2025-03-16:
Hmm I don't think we use any features added since cmake 3.16 so might be able to downgrade that requirement so the system packages work again.

jordancarlin, 2025-12-07:
The install process is the same. Make sure you’re downloading from the `0.20 (Linux Binary Release)` (https://github.com/rems-project/sail/releases/tag/0.20-linux-binary) and not the `0.20` release.

@Alasdair I've been meaning to ask, is there any reason the binaries are part of a separate release?

Alasdair, 2025-12-07:
Oh it's just because it's a separate github action, could probably merge them in future