riscv/sail-riscv #799: Run tests locally failed

trdthg, 2025-03-17:
Can't compile tests locally

test environment

- os: wsl ubuntu24.04
- toolchain: riscv-gnu-toolchain release latest

some log

```
[22/30] Compiling test_hello_world.c
FAILED: test/first_party/rv32d_test_hello_world.c.elf /home/trdthg/repo/sail/sail-riscv/build/test/first_party/rv32d_test_hello_world.c.elf 
cd /home/trdthg/repo/sail/sail-riscv/build/test/first_party && /home/trdthg/riscv/bin/riscv64-unknown-elf-gcc -march=rv32gc -mabi=ilp32d -mno-relax -ffreestanding -nostdlib -static -mcmodel=medany -g -Og -Wall -Werror -T /home/trdthg/repo/sail/sail-riscv/test/first_party/src/common/link.ld -o rv32d_test_hello_world.c.elf /home/trdthg/repo/sail/sail-riscv/test/first_party/src/test_hello_world.c /home/trdthg/repo/sail/sail-riscv/test/first_party/src/common/crt0.S /home/trdthg/repo/sail/sail-riscv/test/first_party/src/common/nanoprintf.c /home/trdthg/repo/sail/sail-riscv/test/first_party/src/common/runtime.c
/home/trdthg/riscv/bin/../lib/gcc/riscv64-unknown-elf/14.2.0/../../../../riscv64-unknown-elf/bin/ld: address 0xa0002010 of rv32d_test_hello_world.c.elf section `.stack' is not within region `if_ram'
/home/trdthg/riscv/bin/../lib/gcc/riscv64-unknown-elf/14.2.0/../../../../riscv64-unknown-elf/bin/ld: rv32d_test_hello_world.c.elf section `.text' will not fit in region `if_ram'
/home/trdthg/riscv/bin/../lib/gcc/riscv64-unknown-elf/14.2.0/../../../../riscv64-unknown-elf/bin/ld: address 0xa0002010 of rv32d_test_hello_world.c.elf section `.stack' is not within region `if_ram'
/home/trdthg/riscv/bin/../lib/gcc/riscv64-unknown-elf/14.2.0/../../../../riscv64-unknown-elf/bin/ld: region `if_ram' overflowed by 12544 bytes
/home/trdthg/riscv/bin/../lib/gcc/riscv64-unknown-elf/14.2.0/../../../../riscv64-unknown-elf/bin/ld: warning: rv32d_test_hello_world.c.elf has a LOAD segment with RWX permissions
collect2: error: ld returned 1 exit status
[24/30] Compiling test_minstret.S
FAILED: test/first_party/rv32d_test_minstret.S.elf /home/trdthg/repo/sail/sail-riscv/build/test/first_party/rv32d_test_minstret.S.elf 
cd /home/trdthg/repo/sail/sail-riscv/build/test/first_party && /home/trdthg/riscv/bin/riscv64-unknown-elf-gcc -march=rv32gc -mabi=ilp32d -mno-relax -ffreestanding -nostdlib -static -mcmodel=medany -g -Og -Wall -Werror -T /home/trdthg/repo/sail/sail-riscv/test/first_party/src/common/link.ld -o rv32d_test_minstret.S.elf /home/trdthg/repo/sail/sail-riscv/test/first_party/src/test_minstret.S /home/trdthg/repo/sail/sail-riscv/test/first_party/src/common/crt0.S /home/trdthg/repo/sail/sail-riscv/test/first_party/src/common/nanoprintf.c /home/trdthg/repo/sail/sail-riscv/test/first_party/src/common/runtime.c
/home/trdthg/riscv/bin/../lib/gcc/riscv64-unknown-elf/14.2.0/../../../../riscv64-unknown-elf/bin/ld: address 0xa0002010 of rv32d_test_minstret.S.elf section `.stack' is not within region `if_ram'
/home/trdthg/riscv/bin/../lib/gcc/riscv64-unknown-elf/14.2.0/../../../../riscv64-unknown-elf/bin/ld: rv32d_test_minstret.S.elf section `.text' will not fit in region `if_ram'
/home/trdthg/riscv/bin/../lib/gcc/riscv64-unknown-elf/14.2.0/../../../../riscv64-unknown-elf/bin/ld: address 0xa0002010 of rv32d_test_minstret.S.elf section `.stack' is not within region `if_ram'
/home/trdthg/riscv/bin/../lib/gcc/riscv64-unknown-elf/14.2.0/../../../../riscv64-unknown-elf/bin/ld: region `if_ram' overflowed by 12628 bytes
/home/trdthg/riscv/bin/../lib/gcc/riscv64-unknown-elf/14.2.0/../../../../riscv64-unknown-elf/bin/ld: warning: rv32d_test_minstret.S.elf has a LOAD segment with RWX permissions
collect2: error: ld returned 1 exit status
[27/30] Building C object c_emulator/CMakeFiles/riscv_sim_rv64d.dir/__/riscv_model_rv64d.c.o
ninja: build stopped: subcommand failed.
```

Timmmm, 2025-03-17:
That's weird, `if_ram` has 512 MB of space so it seems weird that it would overflow by 12 kB.

trdthg, 2025-03-17:
Another small question, where does the riscv-gcc used in the CI come from? I didn't find any install process.

Timmmm, 2025-03-17:
I think CI uses Clang... but actually it probably doesn't work for the release workflow since Clang isn't installed there - I'll have to update that.

Timmmm, 2025-03-17:
I've reproduced this with 13.2.1 - I'll investigate.

PR #802 (2025-03-17): Fix linker script for GNU ld
It seems like lld and ld interpret linker scripts slightly differently. `lld` would put `.bss.mmio` in `if_mmio`, and then when you put `.stack` in `if_ram` it would sensibly reset put it at the start of `if_ram`. However `ld` is a bit dumber and just continues in `if_mmio` leading to overflow.

`ld` also requires manually skipping to the start of `if_ram`, otherwise it tries to allocate starting at 0.

Fixes #799
commit 590131a1 (2025-03-17): Fix linker script for GNU ld

It seems like lld and ld interpret linker scripts slightly differently. `lld` would put `.bss.mmio` in `if_mmio`, and then when you put `.stack` in `if_ram` it would sensibly reset put it at the start of `if_ram`. However `ld` is a bit dumber and just continues in `if_mmio` leading to overflow.

`ld` also requires manually skipping to the start of `if_ram`, otherwise it tries to allocate starting at 0.

Fixes #799
closed 2025-03-17 by Timmmm