riscv/sail-riscv #806: Test can't stop if enavle `-C` option

trdthg, 2025-03-19:
When -C (disable-rvc) is enabled, the test falls into an infinite loop

- cmd: `./build/c_emulator/riscv_sim_rv64d -C ./build/test/first_party/rv64d_test_empty.S.elf > log.txt`
- test asm:
  ```
  .global main
  main:
     ret
  ```
 
log

```
Running file ./build/test/first_party/rv64d_test_empty.S.elf.
ELF Entry @ 0x80002040
CSR mstatus <- 0x0000000A00000000 (input: 0x0000000000000000)
mem[X,0x0000000000001000] -> 0x0297
mem[X,0x0000000000001002] -> 0x0000
[0] [M]: 0x0000000000001000 (0x00000297) auipc t0, 0x0
x5 <- 0x0000000000001000
mem[X,0x0000000000001004] -> 0x8593
mem[X,0x0000000000001006] -> 0x0202
[1] [M]: 0x0000000000001004 (0x02028593) addi a1, t0, 0x20
x11 <- 0x0000000000001020
mem[X,0x0000000000001008] -> 0x2573
mem[X,0x000000000000100A] -> 0xF140
[2] [M]: 0x0000000000001008 (0xF1402573) csrrs a0, mhartid, zero
CSR mhartid -> 0x0000000000000000
x10 <- 0x0000000000000000
mem[X,0x000000000000100C] -> 0xB283
mem[X,0x000000000000100E] -> 0x0182
[3] [M]: 0x000000000000100C (0x0182B283) ld t0, 0x18(t0)
mem[R,0x0000000000001018] -> 0x0000000080002040
x5 <- 0x0000000080002040
mem[X,0x0000000000001010] -> 0x8067
mem[X,0x0000000000001012] -> 0x0002
[4] [M]: 0x0000000000001010 (0x00028067) jalr zero, 0x0(t0)
mem[X,0x0000000080002040] -> 0x006F
mem[X,0x0000000080002042] -> 0x0D40
[5] [M]: 0x0000000080002040 (0x0D40006F) jal zero, 0xd4
mem[X,0x0000000080002114] -> 0x4081
[6] [M]: 0x0000000080002114 (0x4081) c.illegal 0x4081
trapping from M to M to handle illegal-instruction
handling exc#0x02 at priv M with tval 0x0000000000000000
CSR mstatus <- 0x0000000A00001800
within_phys_mem: 0x0000000000000000 not within phys-mem:
  plat_rom_base: 0x0000000000001000
  plat_rom_size: 0x0000000000001000
  plat_ram_base: 0x0000000080000000
  plat_ram_size: 0x0000000080000000
trapping from M to M to handle fetch-access-fault
handling exc#0x01 at priv M with tval 0x0000000000000000
CSR mstatus <- 0x0000000A00001800
within_phys_mem: 0x0000000000000000 not within phys-mem:
  plat_rom_base: 0x0000000000001000
  plat_rom_size: 0x0000000000001000
  plat_ram_base: 0x0000000080000000
  plat_ram_size: 0x0000000080000000
```

jrtc27, 2025-03-19:
Well yeah, the build system passes -march=rv32/64gc (https://github.com/riscv/sail-riscv/blob/9292b9810bec24c4383752d9f61774eb0b9fcfe1/test/first_party/CMakeLists.txt#L103), so if you disable instructions at run time that the assembler was told existed then you should expect things not to work. 0x4081 is `li ra, 0`, corresponding to https://github.com/riscv/sail-riscv/blob/9292b9810bec24c4383752d9f61774eb0b9fcfe1/test/first_party/src/common/crt0.S#L90.

jordancarlin, 2025-03-19:
Once the new configuration system is merged (hopefully soon), there are plans to create tests with different configuration JSON files (either embedded in the test as a comment or as a separate file). In addition to the configuration JSON for the model, these tests will also need to pass the right march string to the compiler.

pmundkur, 2025-06-13:
#1044 could be used to ensure compatible tests are run.