riscv/sail-riscv #812: Simulator no longer builds on macOS

Alasdair, 2025-03-23:
```
/Users/alasdair/repos/sail-riscv/c_emulator/riscv_sim.cpp:66:6: error: no template named 'optional' in namespace 'std'
   66 | std::optional<rvfi_handler> rvfi;
      | ~~~~~^
/Users/alasdair/repos/sail-riscv/c_emulator/riscv_sim.cpp:658:21: error: no member named 'nullopt' in namespace 'std'
  658 |         rvfi = std::nullopt;
      |                ~~~~~^
3 warnings and 2 errors generated.
```
I think we need to specify C++17 somewhere in CMake. I tried setting it in what felt like the obvious place, but that caused issues when building the `.c` files.

jrtc27, 2025-03-23:
Set CMAKE_CXX_STANDARD to 17 and CMAKE_CXX_STANDARD_REQUIRED to On.

Alasdair, 2025-03-24:
Thanks, that fixed my problem!

arichardson, 2025-03-24:
Ah that should have been caught by CI!

arichardson, 2025-03-24:
https://github.com/riscv/sail-riscv/pull/813

jordancarlin, 2025-03-24:
> Ah that should have been caught by CI!

We don’t run CI on macOS right now.

edolnx, 2025-03-24:
I'll look into this and see what I can come up with. Might as well get a Mac Mini setup sooner rather than later to get this working. I can at least start with the older Intel devices I have and then get a Apple Silicon device by end of week.

closed 2025-03-24 by Timmmm