riscv/sail-riscv #850: PMP is not working after the new JSON based configuration

UmerShahidengr, 2025-04-12:
After the recent changes in the sail-model, PMP has stopped working. PMP tests in riscv-arch-test are failing because PMP is not throwing the expected traps. I am sharing a testfile here [pmp64-TOR-S.txt](https://github.com/user-attachments/files/19718687/pmp64-TOR-S.txt) along with its disassembly  [ref_disass.txt](https://github.com/user-attachments/files/19718720/ref_disass.txt) and sail-log [pmp64-TOR-S.log](https://github.com/user-attachments/files/19718724/pmp64-TOR-S.log). 

This test is checking a few pmp cases i.e, setting up a pmp test region, and testing its behavior in various RWX cases like shown below:

```
//Test Case: 2 -- RX Permissions given to the PMP Region. 
// Must give trap on every write access

	LI(x4, PMPREGIONRX)
	csrw pmpcfg0, x4

	RVTEST_GOTO_LOWER_MODE	Smode
	VERIFICATION_RWX	TEST_FOR_EXECUTION

	RVTEST_GOTO_MMODE

//Test Case: 3 -- RW Permissions given to the PMP Region.
// Must give trap on every execute access

	LI(x4, PMPREGIONRW)
	csrw pmpcfg0, x4

	RVTEST_GOTO_LOWER_MODE	Smode
	VERIFICATION_RWX	TEST_FOR_EXECUTION

	RVTEST_GOTO_MMODE

//Test Case: 4 -- R Permissions given to the PMP Region.
// Must give trap on every write & execute access

	LI(x4, PMPREGIONR)
	csrw pmpcfg0, x4

	RVTEST_GOTO_LOWER_MODE	Smode
	VERIFICATION_RWX	TEST_FOR_EXECUTION

	RVTEST_GOTO_MMODE

//Test Case: 5 -- X Permissions given to the PMP Region.
// Must give trap on every read & write access

	LI(x4, PMPREGIONX)
	csrw pmpcfg0, x4

	RVTEST_GOTO_LOWER_MODE	Smode
	VERIFICATION_RWX	TEST_FOR_EXECUTION

	RVTEST_GOTO_MMODE
```

The test is expecting various traps, but it is not giving any trap at all. Spike is working perfectly, whereas the old version of sail is also working, but since sail-model has switched to its new method of JSON based configurations, these tests are not working.
This is just one example code I am sharing in which PMP region is configured in TOR mode, but other modes are also not working (not giving expected traps in the test). Please look into this issue. I am using a default.json script with RISCOF framework to run the test on sail model.

jordancarlin, 2025-04-13:
I dug into this with a git bisect, and it looks like it was actually #773 that broke this, so it is unrelated to the new json config system. Not sure what was wrong in that PR, but it seems to no longer be throwing traps for accesses that should be prohibited by the PMPs. @Timmmm Can you look into this?

Timmmm, 2025-04-14:
Ooop! I am taking a look...

UmerShahidengr, 2025-04-15:
Thanks alot for the quick response

PR #852 (2025-04-15): Fix PMP address matching
In a recent refactor I accidentally removed the * 4 factor in `pmpaddr` registers. This fixes that.

I added a basic test that tests this at least for the NAPOT case. We could do with a more comprehensive test but it's quite difficult to write this sort of test as an ELF. I think maybe this is a situation where unit tests would be appropriate.

Finally I added a missing `sfence.vma` in the `test_max_pmp.c` test (doesn't actually matter for the Sail model but we should be correct), and simplified `crt0.S`'s full match PMP address.

Fixes #850
closed 2025-04-15 by Timmmm