riscv/sail-riscv #860: Configuration validation

jordancarlin, 2025-04-15:
We need some way of ensuring the configuration passed to the model is legal. Right now (almost) all of the `hartSupports` clauses blindly accept the values from the json config file without any validation regarding extensions that require other extensions. We have some of this validation baked into the `currentlyEnabled` clauses, but this will just not enable extensions if the dependencies aren't enabled, which probably isn't very intuitive.

My current thought on this is having a `validateConfiguration` function (or something along those lines) that gets called as part of the init function. It would have a list of asserts that check for valid configurations based on the `hartSupports` clauses (and other relevant configuration parameters). That way the model could exit with a descriptive error that guides the user to fixing their configuration file.

I know we are eventually planning to have a tool that generates these configuration files, but people will inevitably end up modifying them by hand so I think this is still necessary.

@Timmmm @pmundkur @Alasdair any thoughts on this or a better approach?

Timmmm, 2025-04-15:
I think one thing to consider is the formal backends. I may be misremembering this but as I recall they treat the `sys_enable()` callbacks as free functions. I'm not sure if they will for `config` too... If so we might want to have the code legalise the config. For example instead of just blindly believing that the user enabled Sv39 if they enabled Sv48, we should have the `hartSupports(Ext_Sv48)` check both. So that no matter what values the `config` has it will result in a legal configuration.

The big downside there is that it's going to be really confusing when you accidentally enable Sv48 but not Sv39, and then the model silently disables both. So we probably *also* want the validation function. I think your proposal to call it at init time and fill it with asserts is a good one.

@Alasdair @bacam can you let us know if I've totally misremembered how the free functions stuff works?

jordancarlin, 2025-04-15:
I believe we pass a configuration file to Sail when compiling for one of the theorem provers and these values are all set at compile time.

pmundkur, 2025-04-15:
I'd prefer to keep minimal dependency checking in the Sail code itself, to roughly what is there now, and to reflect what's in the ISA spec.  The rest of the configuration validation (or automatic enabling of extension dependencies etc) should be in the `c_emulator` or a helper script called by the `c_emulator` if needed.  This script can also generate the RVFI version of the config, so that RVFI could be made to work again that way.

It also helps to avoid duplicating validation logic in the Sail code and in the helper script. The  script could also be more helpful about explaining what's wrong with the config than the Sail model could.  There should be no way of using a config without it going through the script.

One reason to have a script outside of `c_emulator` is so that it can be reused for the other backends.

So I'd rather not have a `validationConfiguration` function in Sail, but have it in `c_emulator` or a script called by it.  But I'm open to being persuaded otherwise.

Timmmm, 2025-04-15:
I dunno I think it's better in the Sail code precisely so it can easily be used by all the back-ends. Plus it means you're closer to where it will be actually used, and you get to use Sail which is a fair bit nicer than C.

I can't really see any advantages of doing it in C or a separate script?

jordancarlin, 2025-04-15:
I think there are 2 aspects to this. 

One is creating the json configuration file. I think developing scripts/adapting existing tools (like riscv-config or UDB) to do this would be great and is definitely something we'll need for the Sail configuration to be widely adopted. These tools will probably do their own validation, so the json file that they output should (in theory) not have any issues.

It is quite likely that people will end up manually modifying the output json file though, so we still need some level of validation directly connected to Sail (either in the model itself or in the various backends) to avoid confusing errors/unexpected behavior. 

I'd advocate for this part being in Sail (as opposed to the c_emulator or another script that the c_emulator calls) for a few reasons. As @Timmmm already mentioned, having the validation directly in Sail means that all of the backends get it for free. There is no additional work (especially from a maintenance perspective as new things are added) to support the various different backends. Beyond that, many of the constraints that I'm envisioning in this configuration check are constraints imposed directly by the ISA. For example, a hart cannot support the F extension unless it supports the Zicsr extension. There are numerous dependencies like this scattered throughout the model, and I think if we are viewing Sail as the executable reference of the spec then those requirements should be in Sail (and having these dependencies in a single, easy to find location is a nice benefit too). As much as possible, the c_emulator should deal with platform and other non isa aspects that are needed to run the model but are not a part of the spec itself while directly from the spec should be in the Sail.

pmundkur, 2025-04-15:
The main concern I have with doing it in Sail is the user experience.  Currently the config loader in the Sail runtime just asserts on a parse failure with no indication on where the error was.    Sail has also very limited support for printing and data types like dictionaries.   A script would be able to do better error reporting.

The Sail compiler can generate a JSON schema for the config.  The config can be validated against the schema using a script, but not within the Sail model.

Sail is a great ISA DSL, but it is not a general purpose language.  It seems odd to do JSON config validation and error reporting in it.

I think it would be possible to do some validation in Sail.  But if we're going to have a script anyway for schema validation and a better user experience, we'd end up duplicating the validation logic and make it possible for the Sail external and Sail internal validations to get out-of-sync.

jordancarlin, 2025-04-16:
> I agree we should validate against the schema outside of Sail, but I think that's a separate kind of validation to what [@jordancarlin](https://github.com/jordancarlin) is talking about since the schema doesn't capture everything (e.g. mutually exclusive extensions).
> 
> I think we want both:
> 
> 1. C emulator validates the JSON against the schema.
> 2. Sail validates that the settings are legal.

Yes, that’s exactly what I’m thinking. I don’t think either is sufficient on its own and the combination of the two will be able to give the most actionable error messages.

jordancarlin, 2025-04-21:
Expanding on this from discussion at today's the TGMM, there is also the issue of extensions like Zama16b, which are really just constraints on architectural options.

Zama16b mandates that "misaligned loads, stores, and AMOs to main memory regions that do not cross a naturally aligned 16-byte boundary are atomic". We separately have a numerical configuration parameter for the allowed misaligned access size that is handled atomically. #861 implements both of these as options and silently overrides the numerical parameter if it conflicts with Zama16b, which seems potentially confusing for users.

There are two approaches that I can think of for this (which were briefly discussed at today's meeting):
- Just have the numerical parameter. If an implementation supports Zama16b, then they need to set that value to 16. Maybe have a comment for that parameter clarifying this.
- Have Zama16b and the numerical parameter as separate options, and in the configuration validation throw an error if Zama16b is supported but the misaligned `allowed_within` parameter is set to less than 16.

There are a number of other parameters/extensions that this same issue applies to.

Timmmm, 2025-04-21:
I definitely don't think we should have profile options that override any settings. That's just very confusing.

What we could do is add some option that verifies that you do *comply* with profile options (including entire profiles).

Something like:

```
{
  "verify_profiles": {
    "RVA22": true,
    "Zama16": true,
   }
```

Then in our Sail function that validates the config we check if the actual settings comply with those profiles.

It would be good to see some actual demand for this before we implement it. Kind of feels like it might not be really necessary. I suspect users would be happy with just having an example `RVA22.json`.

pmundkur, 2025-04-22:
We should collect the various items to validate in one place; we could use this issue for now.   In addition to the above,

- ensuring `memory.misaligned.allowed_within` is a power of 2 (from #861)
- VLEN >= ELEN , VLEN <= 2^16

allenjbaum, 2025-04-22:
There are some other requirements that you can glean from the riscv-config
code as well (e.g. D without F, etc.)
MXLEN is now readonly, and must be>SXL./UXL, etc

pmundkur, 2025-06-13:
@jordancarlin do you think we can close this now that we have the initial framework for validation?

closed 2025-06-14 by jordancarlin