riscv/sail-riscv #874: Enforce additional GitHub checks on PRs

jordancarlin, 2025-04-21:
We've had 2 PRs get merged recently that broke CI (see #826 and #873). One that added a test but didn't have the new config changes and one that added a file that a new pre-commit hook checks. While both of these ended up being straightforward to fix, it's still not great to be merging PRs that break CI.

One easy solution that would have avoided both of these issues is requiring PR branches to be up to date with master before merging them. GitHub can enforce this for PRs if we want. See https://docs.github.com/en/repositories/configuring-branches-and-merges-in-your-repository/managing-protected-branches/about-protected-branches. It probably makes sense to at least enable “Require branches to be up to date before merging” and maybe also “Require status checks before merging”. @pmundkur thoughts on this?

Timmmm, 2025-04-21:
If we're going to do that it would be better to just enable the merge queue.  I think we should do that. As far as I know there are no downsides that would really affect us.

pmundkur, 2025-04-21:
I agree with @Timmmm that a merge queue seems to be a good solution.

jordancarlin, 2025-04-21:
My only concern with a merge queue is that it looks like they only allow one kind or merging globally (either squash, rebase, or merge commit). We tend to pick different strategies depending on the PR (squash for most PRs, merge/rebase for PRs with several structured commits like some of the recent refactorings). It seems like this would force us to either split these PRs up and enable squashing for all the PRs or require PR authors to clean up the git history for their PRs and rebase them all. Either way, it creates some additional work for developers.

pmundkur, 2025-04-22:
Wouldn't it be additional work for the PR author either way: either require the PR branch to be up to date, or require the author to clean up the git history before it gets added to the merge queue?  Without a clear guideline on when a PR will be merged, the PR author will be left to continually keep the branch up to date even with a `will-be-merged` tag.
The global setting for the merge queue is unfortunate though.  The PR review could specify what cleanup needs to be done before it gets added to the queue.  Would that work?

Timmmm, 2025-04-22:
> they only allow one kind or merging globally

Hmm yeah that is rather unfortunate. I found [a request](https://github.com/orgs/community/discussions/68723) for this feature but there's been no response.

I don't think it would be a huge loss to squash all PRs though so maybe we try that? If we *reaaaally* care about the history then maybe it should be separate PRs anyway (admittedly a workflow that is poorly supported by Github).

jordancarlin, 2025-04-23:
> I don't think it would be a huge loss to squash all PRs though so maybe we try that? If we _reaaaally_ care about the history then maybe it should be separate PRs anyway (admittedly a workflow that is poorly supported by Github).

Yeah, I think I agree that squashing seems like the best choice if we have to choose one.

pmundkur, 2025-05-05:
Merge queue has been enabled for the main branch today.

jordancarlin, 2025-05-06:
Closing now that the merge queue is enabled and working.

closed 2025-05-06 by jordancarlin