riscv/sail-riscv #903: Generating SMT files

algebravic, 2025-04-29:
I'm trying to generate SMT output from the RISCV specs. I've cloned the repository, made a build subdirectory and then do
 `cmake -DCMAKE_BUILD_TYPE=Release ..` and then did `make generated_smt_rv32d`. I get the output

```
make generated_smt_rv32d
-- Found sail: /Users/victorsmiller/.opam/default/bin/sail
-- Sail library directory: /Users/victorsmiller/.opam/default/share/sail
-- Configuring done (0.1s)
-- Generating done (0.2s)
-- Build files have been written to: /Users/victorsmiller/Programming/Circuits/sail-riscv/build
Building smtlib2 from Sail model (rv32d)
Built target generated_smt_rv32d
```

But after this the only `smt2` file that I find in the build directory is `model_cmulr_equivalence.smt2` which only looks like a small snippet of the architecture. Is there something else that I need to do?

pmundkur, 2025-04-29:
That build target builds the formulae for the properties specified in the model
(search for  `$[property]`).   There is currently only one property defined, the one for `cmulr` you found.  There will be a couple more once we merge #861.

Was there anything in particular you were looking to do?  We'd welcome adding more useful properties!

algebravic, 2025-04-29:
Thanks for explaining that. I have a somewhat ambitious project connected
with building a zkVM (zero knowledge virtual machine). Right now we've
specified constraints by hand to do the zk emulation of the RISCV 32d. I
wanted to take the SAIL specs and have it generate SMT so that I could
check if the hand generated constraints are compatible with the spec. We
have everything from instruction decode to emulation of all of the
operations.