riscv/sail-riscv #913: Add performance benchmarking to CI

Timmmm, 2025-05-01:
Following #889 we should boot Linux in CI workflows and output performance stats in the CI results.

When I tested just now I only get about 30 kIPS, whereas when I tested for the RISC-V summit last year I was getting something like 300 kIPS. I'll try and figure out why, but it would be good to learn about these things straight away.

Timmmm, 2025-05-02:
Hmm seems like it's basically all in the PMP checks for fetch.

I think the issue is that accessing a `vector()` *always* uses GMP even if the vector length is known to the compiler. See [this example](https://godbolt.org/z/dqqbbh3fG), so the regression was caused by 4de2bff12d967d91dd064e4a49e25ca4785f25e3 which changed the PMP registers to a vector.

```
uint64_t zaccess(int64_t zi)
{
  __label__ end_function_2, end_block_exception_3;

  uint64_t zcbz30;
  {
    sail_int zgsz34;
    CREATE(sail_int)(&zgsz34);        <--------This is the slow bit. sail_int is a libgmp integer.
    CONVERT_OF(sail_int, mach_int)(&zgsz34, zi);
    zcbz30 = vector_access_zz5vecz8z5bv8z9(zarray, zgsz34);
    KILL(sail_int)(&zgsz34);
  }
end_function_2: ;
  return zcbz30;
end_block_exception_3: ;

  return UINT64_C(0xdeadc0de);
}
```

Timmmm, 2025-05-02:
Seems to me `vector_access` etc. should just take a `size_t`? It's not like you can actually have `vector(10000000000000000000000000000, ...`.

Timmmm, 2025-05-02:
Ah interesting, so the reason it takes a `sail_int` is that it uses this type declaration:

```
val plain_vector_access = pure {
  ocaml: "access",
  interpreter: "access",
  lem: "access_list_dec",
  coq: "vec_access_dec",
  lean: "GetElem?.getElem!",
  _: "vector_access"
} : forall ('n : Int) ('m : Int) ('a : Type), 0 <= 'm < 'n. (vector('n, dec, 'a), int('m)) -> 'a

overload vector_access = {bitvector_access, plain_vector_access}
```

Interestingly there's a similar declaration for bitvector access:

```
val bitvector_access = pure {
  ocaml: "access",
  interpreter: "access",
  lem: "access_vec_dec",
  coq: "access_vec_dec",
  lean: "BitVec.access",
  _: "vector_access"
} : forall ('n : Int) ('m : Int), 0 <= 'm < 'n . (bits('n), int('m)) -> bit
```

Which means that accessing a single bit from a bit vector like `foo[15]` is *also* very slow. If you access a range like `foo[15 .. 15]` then it uses a different function which doesn't have this problem.

I guess we can add a constraint...

```
val plain_vector_access = pure {
...
  _: "vector_access"
} : forall ('n : Int) ('m : Int) ('a : Type), 0 <= 'm < 'n & 'n < 2 ^ 64. (vector('n, dec, 'a), int('m)) -> 'a
```

which presumably will make it use `uint64_t`, but it's also going to break any code that doesn't have similar constraints. I guess the best solution is something like this:

```
val plain_vector_access = pure {
  ocaml: "access",
  interpreter: "access",
  lem: "access_list_dec",
  coq: "vec_access_dec",
  lean: "GetElem?.getElem!",
  _: "vector_access"
} : forall ('n : Int) ('m : Int) ('a : Type), 0 <= 'm < 'n. (vector('n, dec, 'a), int('m)) -> 'a

val plain_vector_access_u64 = pure {
  ocaml: "access",
  interpreter: "access",
  lem: "access_list_dec",
  coq: "vec_access_dec",
  lean: "GetElem?.getElem!",
  c: "vector_access_u64",
  _: "vector_access",
} : forall ('n : Int) ('m : Int) ('a : Type), 0 <= 'm < 'n & 'n < 2 ^ 64. (vector('n, dec, 'a), int('m)) -> 'a

overload vector_access = {bitvector_access_u64, bitvector_access, plain_vector_access_u64, plain_vector_access}
```

Might have a go if I get a chance.

@Alasdair FYI

trdthg, 2025-05-20:
> I think the issue is that accessing a vector() always uses GMP even if the vector length is known to the compiler. 

I previously did a flame graph analysis on sail-riscv

[image]
> seems github removed the .svg ext, you can't click on it, but you can download it

You can see that various gmp operations take up most of the cpu time (more than 50%)

pydrofoil also made many optimizations in this part