riscv/sail-riscv #925: Rework AccessType

Timmmm, 2025-05-07:
In addition to #787, I think there are some more things we should change about `AccessType`.

1. There are some `aq`, `rel`, `res`, `con` flags that are really part of the access type, but they're in separate bool flags for some reason.
2. `ReadWrite` is only used for AMOs so it should be called `AMO`.
3. `AMO` should include the AMO type. This is needed for PMA checks.
4. There should be `LoadReserved` and `StoreConditional` types.

I think in the end we'll have something like

```
union AccessType ('a : Type) = {
  Read             : 'a,
  Write            : 'a,
  // operation, Read, Write
  AMO              : (amoop, 'a, 'a),
  // acquire, release
  LoadReserved     : (bool, bool, 'a),
  StoreConditional : (bool, bool, 'a),
  InstructionFetch : unit,
  Cache            : CacheAccessType,
  // For cm.jt
  JumpTableFetch   : unit,
}
```

The purpose of the `'a` by the way is to allow CHERI to mark accesses as storing tagged capabilities or loading possibly-tagged capabilities. We might be able to remove it once CHERI is fully integrated (haven't really thought about that too much yet though).

jordancarlin, 2025-05-07:
We might want something more generic for `JumpTableFetch` that just indicates it is a non instruction fetch with execute access so that it can be used for other similar accesses down the road, but otherwise this seems like a good list.

trdthg, 2025-05-09:
what about amocas? it also may have `aq|rl`

trdthg, 2025-05-09:
```
// operation, Read, Write
  AMO              : (amoop, 'a, 'a),
```

Why AMO doesn't have acquire and release?

Timmmm, 2025-05-09:
Ah you're right; it should do.

pmundkur, 2025-11-18:
For Zicfiss, #377 added additional `SSRead/SSWrite/SSReadWrite` access types:
https://github.com/ved-rivos/sail-riscv-cfi/blob/161506364fe6d548ef7c9af8caa66e19a7eab890/model/riscv_types.sail#L136

Instead of replicating the `Load/Store/LoadStore` again for shadow stack access, we could use the `'a` parameter,  and have `'a` instantiated with `Data` or `ShadowStack`.

I'm not sure how that interacts with CHERI.  That spec (if I'm reading the right place) currently says Zicfiss is incompatible (so far):
https://github.com/riscv/riscv-cheri/blob/d1c9e92ce47f4d12890503c190dde1990d1a9d91/src/cheri/introduction.adoc?plain=1#L41

Does that mean that there would be no need for `ShadowStackCap` or `ShadowStackTagged`, and hence allowing `'a` to be instantiated by `Data/ShadowStack/{Cap or Tagged}`  when CHERI is integrated?

Or would it be better to use `SSRead : 'a/SSWrite : 'a/SSReadWrite : ('a,'a)` as in the PR?