riscv/sail-riscv #938: Add config option for tail/mask agnostic behaviour

Timmmm, 2025-05-13:
Tail/mask agnostic elements can either be left undisturbed or written with ones. Currently the code always uses the undisturbed option:

```
      result[i] = match tail_ag {
        UNDISTURBED => vd_val[i],
        AGNOSTIC    => vd_val[i] /* TODO: configuration support */
      }
```

The range of possibilities is, as usual, basically infinite:

> When a set is marked agnostic, the corresponding set of destination elements in any vector destination operand can either retain the value they previously held, or are overwritten with 1s. Within a single vector instruction, each destination element can be either left undisturbed or overwritten with 1s, in any combination, and the pattern of undisturbed or overwritten with 1s is not required to be deterministic when the instruction is executed with the same inputs.

However there are a few sensible options to support. Hard to know what people will need until they ask for it, so I think we should go with an enum here instead of a bool for future expandability.

We probably want at least the first two of these. I imagine some architectures could do different things for registers that are fully within the [tail or masked](https://riscv-specs.timhutt.co.uk/spec/20250508/riscv-unprivileged-20250508.html#sec-inactive-defs).

```
enum VectorAgnosticBehavior = {
   // Always leave elements with their previous value.
   AlwaysUndisturbed,
   // Always overwrite elements with ones.
   AlwaysOnes,
   // Write ones if the elements are not in a register entirely within the tail.
   OnesUnlessTailRegister,
   // Write ones if the elements are not in a register entirely within the tail, or entirely masked.
   OnesUnlessInactiveRegister,
}
```

But let's start with the first two until someone needs more.

allenjbaum, 2025-05-13:
> Also, since the behaviour can be quite complicated (he says, after
> checking our specs), we might want to put this in a function like
>
> function is_undisturbed(element_index, vstart, vl, ma, masked, is_tail_reg, ...) -> bool = {
>   .. simple standard implementation ...
> }
>
> so that it can easily be replaced for implementations that do things like
> special casing vl = 0, or vstart >= vl, or vstart = 0, etc.

I think undisturbed is pretty definite.
Agnostic is open-ended. There is no way to test all (virtually infinite)
possibilities, and OOO implementations could vary between undisturbed and
agnostic on an element by element and instance by instance basis. The only
way I've come up for testing agnostic is to compare against undistrubed,
and if it miscompares, check that all the miscompared elements are all1.
Effectively, Sail never has to worry about case 2. Tests have to
accommodate that, Sail doesn't.
There may be some CI tests that want to play with overwrite with ones, but
otherwise I can see where it would be necessary.

Timmmm, 2025-05-14:
Right but some people use Sail as an exact golden model so we want to make it as easy as possible for them to match the behaviour of Sail to their design. Ideally using the config system, failing that by making patching as uninvasive as possible.

People may also want to use it to test software with a variety of possible implementations so they don't inadvertantly rely on implementation defined behaviour.

allenjbaum, 2025-05-14:
Your last reason is a good one, but it will interfere with test unless you
have a separate configuration option beyond agnostic vs undisturbed  (e.g.
agnostic_ones.aganostic_undisturbed)

radimkrcmar, 2025-05-15:
For the software testing purposes, it would be nice to have a configuration that taints the agnostic values and emits a warning if software eventually uses them in an observable manner.
A single pass would detect software that uses implementation defined behavior, but we'd have to handle the taint in all vector code.

Timmmm, 2025-05-15:
Nice idea but I think that's not really possible in general, or at least it's not something the Sail model could reasonably do. E.g. you could store the results to memory and then load them much later, so then you have to taint all memory. It's also not that uncommon that temporarily "unknown" values get squashed to a single known value but in a way that isn't clear to tools - this is quite a common problem with Verilog's `x` values. I'm not a symbolic execution expert though so maybe it would work.

> but it will interfere with test unless you
have a separate configuration option beyond agnostic vs undisturbed  (e.g. agnostic_ones.aganostic_undisturbed)

I don't follow this. What would this extra option do?

allenjbaum, 2025-05-16:
Sail has to decode a vector op and determine if it is undisturbed or
agnostic. The extra option parameter tells Sail to handle it as all-one or
undisturbed ( but only for all masked elements at the same time, which
isn’t a requirement.