riscv/sail-riscv #951: Lean build broken after sfence_vma_illegal_if_svbare_only config option added

jordancarlin, 2025-05-14:
Lean compilation was working when I opened #941, but seems like it has broken again. It's throwing the following error:

```
Expression E_config config extensions.Svbare.sfence_vma_illegal_if_svbare_only not translatable yet.
```

@ineol @tobiasgrosser

Alasdair, 2025-05-14:
Could there be a spelling mistake somewhere (so the model and config file don't see the same option)? I think it's set up to statically instantiate the model from the default config, so this shouldn't appear when generating Lean.

jordancarlin, 2025-05-15:
Just took a look, and it seems to be complaining because the config is used in the `when` guard.

Changing

```
mapping clause encdec = SFENCE_VMA(rs1, rs2)
  <-> 0b0001001 @ encdec_reg(rs2) @ encdec_reg(rs1) @ 0b000 @ 0b00000 @ 0b1110011
  when virtual_memory_supported() | not(config extensions.Svbare.sfence_vma_illegal_if_svbare_only : bool)
```

to 

```
let sfence_vma_illegal_if_svbare_only : bool = config extensions.Svbare.sfence_vma_illegal_if_svbare_only

mapping clause encdec = SFENCE_VMA(rs1, rs2)
  <-> 0b0001001 @ encdec_reg(rs2) @ encdec_reg(rs1) @ 0b000 @ 0b00000 @ 0b1110011
  when virtual_memory_supported() | not(sfence_vma_illegal_if_svbare_only)
```
seems to fix it.

We could update it in the model, but I assume this is a Sail compiler bug. @Alasdair

Alasdair, 2025-05-15:
Ah, yes that should work. I'll fix it.

jordancarlin, 2025-05-15:
This was fixed with https://github.com/rems-project/sail/pull/1301 and (after flushing the Sail cache for the repo) Sail now successfully outputs the Lean files.

Lean still fails to build the output, but that's a separate issue that I think will be fixed by https://github.com/rems-project/sail/pull/1303.

closed 2025-05-15 by jordancarlin