riscv/sail-riscv #985: Seeking help for implementing the Sail model for a new RISC-V extension.

ybc-alkaid, 2025-05-27:
Hi maintainers,

I'm from the RISC-V SPMP (S-mode Physical Memory Protection) Task Group. We need to implement the Sail model before the specification freeze deadline, but our team lacks Sail expertise. Would it be appropriate to request help and/or guidance from the Sail modeling community? Any suggestions would be greatly appreciated.

Some pointers:
SPMP mailing list: https://lists.riscv.org/g/tech-spmp
SPMP specification repo: https://github.com/riscv/riscv-spmp/releases/tag/v0.9.10

ybc-alkaid, 2025-05-30:
I tried to implement SPMP in sail by mimicking the PMP (It was an old version of SPMP, which has a similar function to PMP) :
https://github.com/riscv/sail-riscv/compare/master...ybc-alkaid:sail-riscv:smpu
Would you please check whether I did it right?

Other questions: 
- The SPMP supports indirect CSR access. Do I have to implement it in Sail or just the logic of SPMP itself?
- The SPMP shares hardware resources (some CSRs and the matching logic) with the PMP. E.g., there are 64 PMP/SPMP rules in total, and we use a CSR to determine how many rules are delegated to SPMP (let's say 32). Then the PMP can access the hardware resources of first 32 rules and the SPMP can access the rest. How do I describe something like this in Sail? 

Thank you!

allenjbaum, 2025-05-31:
> master...ybc-alkaid:sail-riscv:smpu

SPMP is not currently ratified, but the proposal has *optional* indirect
CSR  says:
2.10. Access Methods of SPMP CSRs
How SPMP CSRs are accessed depends on whether the Sscsrind extension is
implemented or not.

Direct CSR access: SPMP CSRs can be accessed directly with corresponding
CSR numbers if the Sscsrind extension is not implemented.

Indirect CSR access: The SPMP supports indirect CSR access if the Sscsrind
extension is implemented. E
ach combination of siselect and sireg represents an access to the
corresponding SPMP CSR.

So, it can be implemented with direct CSR address only, or indirect only.

jordancarlin, 2025-05-31:
We should implement all of the functionally of the extension in Sail though. #956 adds support for the indirect CSR access extensions, so it probably makes sense to wait on that to be merged before tackling the indirect CSR access part of SPMP.