riscv/sail-riscv #1002: Some assembly clauses missing explicit SP operand

s-khechnev, 2025-06-04:
While the RISC-V Instruction Set Manual is not really an assembly language reference, there are some examples here and there which I presume show preferred implementations of assembly syntax. The assembly definitions of instructions `c.ldsp`, `c.sdsp` are not defined with `(sp)`. But in LLVM and RISC-V ISA (Version 20250508) there are examples which include it.

Section 34.5.3
```
c.ldsp rd, offset(sp)
```

Section 34.5.4
```
c.sdsp rs2, offset(sp)
```

Assembly clauses:
```
mapping clause assembly = C_LDSP(uimm, rd)
  <-> "c.ldsp" ^ spc() ^ reg_name(rd) ^ sep() ^ hex_bits_6(uimm)
  when rd != zreg & xlen == 64

mapping clause assembly = C_SDSP(uimm, rs2)
  <-> "c.sdsp" ^ spc() ^ reg_name(rs2) ^ sep() ^ hex_bits_6(uimm)
  when xlen == 64
```

Also.
In LLVM, instructions  [`c.addi16sp`](https://github.com/llvm/llvm-project/blob/54d544b83141dc0b20727673f68793728ed54793/llvm/test/MC/RISCV/rv32c-only-valid.s#L33), [`c.addi4spn`](https://github.com/llvm/llvm-project/blob/54d544b83141dc0b20727673f68793728ed54793/llvm/test/MC/RISCV/rvc-valid.s#L105), [`c.fldsp`](https://github.com/llvm/llvm-project/blob/54d544b83141dc0b20727673f68793728ed54793/llvm/test/MC/RISCV/rv32dc-valid.s#L22), [`c.flwsp`](https://github.com/llvm/llvm-project/blob/54d544b83141dc0b20727673f68793728ed54793/llvm/test/MC/RISCV/rv32fc-valid.s#L29), [`c.fsdsp`](https://github.com/llvm/llvm-project/blob/54d544b83141dc0b20727673f68793728ed54793/llvm/test/MC/RISCV/rv32dc-valid.s#L27), [`c.fswsp`](https://github.com/llvm/llvm-project/blob/54d544b83141dc0b20727673f68793728ed54793/llvm/test/MC/RISCV/rv32fc-valid.s#L35), [`c.lwsp`](https://github.com/llvm/llvm-project/blob/54d544b83141dc0b20727673f68793728ed54793/llvm/test/MC/RISCV/rvc-valid.s#L31), [`c.swsp`](https://github.com/llvm/llvm-project/blob/54d544b83141dc0b20727673f68793728ed54793/llvm/test/MC/RISCV/rvc-valid.s#L35) explicitly include the `SP` operand.
But the assembly clauses of these instructions do not include the `SP` operand.
```
mapping clause assembly = C_ADDI16SP(imm)
  <-> "c.addi16sp" ^ spc() ^ hex_bits_signed_6(imm)
  when imm != 0b000000

mapping clause assembly = C_ADDI4SPN(rdc, nzimm)
      if nzimm != 0b00000000
  <-> "c.addi4spn" ^ spc() ^ creg_name(rdc) ^ sep() ^ hex_bits_10(nzimm @ 0b00)
      if nzimm != 0b00000000

mapping clause assembly = C_FLDSP(uimm, rd)
      if (xlen == 32 | xlen == 64)
  <-> "c.fldsp" ^ spc() ^ freg_name(rd) ^ sep() ^ hex_bits_6(uimm)
      if (xlen == 32 | xlen == 64)

mapping clause assembly = C_FLWSP(imm, rd)
  <-> "c.flwsp" ^ spc() ^ freg_name(rd) ^ sep() ^ hex_bits_6(imm)
  when xlen == 32

mapping clause assembly = C_FSDSP(uimm, rs2)
      if (xlen == 32 | xlen == 64)
  <-> "c.fsdsp" ^ spc() ^ freg_name(rs2) ^ sep() ^ hex_bits_6(uimm)
      if (xlen == 32 | xlen == 64)

mapping clause assembly = C_FSWSP(uimm, rs2)
  <-> "c.fswsp" ^ spc() ^ freg_name(rs2) ^ sep() ^ hex_bits_6(uimm)
  when xlen == 32

mapping clause assembly = C_LWSP(uimm, rd)
  <-> "c.lwsp" ^ spc() ^ reg_name(rd) ^ sep() ^ hex_bits_6(uimm)
  when rd != zreg

mapping clause assembly = C_SWSP(uimm, rs2)
  <-> "c.swsp" ^ spc() ^ reg_name(rs2) ^ sep() ^ hex_bits_6(uimm)
```

jordancarlin, 2025-06-06:
@ThinkOpenly Any idea why this didn't pop up when doing the assembly cross checking between UDB and Sail? Are they both wrong?

Timmmm, 2025-06-06:
> Does Sail even need to maintain its assembly clauses? (Probably a discussion for another day.)

Yeah they're extremely useful as a disassembler. I guess we could have tests by comparing them with objdump's output, or by trying to assemble the output. The latter probably makes more sense because we only really care that it is valid; not that it *exactly* matches objdump.

Something like:

1. Loop through a load of opcodes. Maybe all 1 billion of them? We can skip illegal ones. Not sure how many that leaves.
2. Use Sail to generate the assembly for them, and write it to a file.
3. Assemble it with `as`.
4. Read the file and confirm the opcodes match what we expect.

closed 2025-06-13 by Timmmm