riscv/sail-riscv #1008: Verify disassembly

Timmmm, 2025-06-06:
See [this idea](https://github.com/riscv/sail-riscv/issues/1002#issuecomment-2948711981). We can loop through all opcodes, disassemble them with Sail then reassemble them with GAS and verify we get the same result. I have code for this and have found a few bugs so far (PRs incoming...).

Timmmm, 2025-06-06:
Full `.s` file ended up being 13GB, with 380,889,914 legal instructions (out of about 1 billion).

Timmmm, 2025-06-06:
And it takes 1h26m to generate. Assembling it looks like it is *much* faster; just a few minutes.

pmundkur, 2025-06-06:
This is awesome!  It would be great to add this tooling to the repo so that we can keep checking the model as we add extensions.

arichardson, 2025-06-06:
This is really nice! I wonder if it would be quicker to write a .sail file that loops over all opcodes (skipping custom space, etc.) and prints the output to a file instead of using an input binary file?

Timmmm, 2025-06-07:
> This is really nice! I wonder if it would be quicker to write a .sail file that loops over all opcodes (skipping custom space, etc.) and prints the output to a file instead of using an input binary file?

That's what I have done. It prints the disassembly and the opcode in a comment for each instruction. Then I'm planning to assemble that file, then objcopy it to a bare binary, then write a script to check the contents match the opcode comments.

Haven't quite got past GAS errors and `assembly` mapping bugs yet though...

I'll push the code next week.

Timmmm, 2025-06-25:
WIP branch: https://github.com/Timmmm/sail-riscv/tree/user/timh/all_legal_assembly