riscv/sail-riscv #1014: Reserved fence handling is not quite right

Timmmm, 2025-06-06:
We have this code:

```
/* ****************************************************************** */
/* The reserved fences are not really hints, but for now they
 * are in this file to be also treated as nops to avoid trapping.
 */

union clause ast = FENCE_RESERVED : (bits(4), bits(4), bits(4), regidx, regidx)

mapping clause encdec = FENCE_RESERVED(fm, pred, succ, rs, rd)
  <-> fm : bits(4) @ pred : bits(4) @ succ : bits(4) @ encdec_reg(rs) @ 0b000 @ encdec_reg(rd) @ 0b0001111
      when (fm != 0b0000 & fm != 0b1000) | rs != zreg | rd != zreg

function clause execute (FENCE_RESERVED(fm, pred, succ, rs, rd)) = RETIRE_SUCCESS
```

However this isn't quite right on a few fronts. The spec says:

> The unused fields in the FENCE instructions--rs1 and rd--are reserved for finer-grain fences in future extensions. For forward compatibility, base implementations shall ignore these fields, and standard software shall zero these fields. Likewise, many fm and predecessor/successor set settings are also reserved for future use. Base implementations shall treat all such reserved configurations as FENCE instructions (with fm=0000), and standard software shall use only non-reserved configurations.

Therefore:

1. They should *ignore* `rs` and `rd`, not check that they are non-zero.
2. The execute clause shouldn't be a nop - it should delegate to `FENCE`, or do the same thing as it (call `sail_barrier()`).
3. Even if `fm == 0b1000` the instruction can still be reserved if it isn't `rw, rw`. In fact it should be executed with `fm = 0b0000`.

This isn't too critical in the Sail model since fences don't really do anything, but I suspect one day we will want to expose them to some kind of verification callback / logging interface and then they should be correct.

jrtc27, 2025-06-06:
The assembly syntax should be 0 for the set

Timmmm, 2025-06-07:
Yeah I guessed that too but [it doesn't work unfortunately](https://godbolt.org/z/6GP774sPb). :-/

Timmmm, 2025-06-07:
Ah interesting. Probably makes sense to use it then even if GAS doesn't support it. It would be nice if there was a proper specification for RISC-V assembly!

Do you have a feel for whether LLVM is more reliable / up-to-date with RISC-V support than binutils? I ran into a couple of other issues like it doesn't support Zabha (half/byte AMOs) yet, and also it told me I needed to enable Zvkb even after I had... Kind of weird.