riscv/sail-riscv #1016: Predicate `currentlyEnabled()` on `mstatus.{UXL,SXL,MXL}` instead of `xlen` where relevant

pmundkur, 2025-06-07:
This is more accurate and helps eventually support writable UXL/SXL in the future.  For most extensions, the XL for `cur_privilege` would apply, but for others it won't.
See  https://github.com/riscv/sail-riscv/pull/969#discussion_r2132506357 and https://github.com/riscv/sail-riscv/pull/969#discussion_r2133209557

radimkrcmar, 2025-06-07:
Sounds good.  `mstatus.MXL` was probably meant to be `xlen`.

Would you accept patches that rename `xlen` to `mxlen`?
I think it would be very confusing if we named something else as XLEN and used `xlen` as MXLEN.

pmundkur, 2025-06-09:
> Would you accept patches that rename `xlen` to `mxlen`? I think it would be very confusing if we named something else as XLEN and used `xlen` as MXLEN.

That makes sense.  `XLEN` is now inherently a dynamic entity (whereas MXLEN is defined as a constant in recent specs), so it makes sense to avoid using it in our definitions and configuration.