riscv/sail-riscv #1018: Rename `ast` type to `instruction`

Timmmm, 2025-06-07:
I presume `ast` means Abstract Syntax Tree, which isn't at all what it is. It's just a decoded instruction. We should rename the type and variables to `instruction` (or maybe `Instruction`? I dunno if our style is really consistent).

jrtc27, 2025-06-07:
Eh, the current Sail documentation says:

> A Sail specification will typically define an abstract syntax type/tre (AST) of machine instructions, a decode function that takes binary values to AST values, and an execute function that describes how each of those behaves at runtime, together with whatever auxiliary functions and types are needed.

So this would diverge from what the language manual says you should do.

pmundkur, 2025-06-09:
We discussed this this morning, and agreed that `instruction` is clearer.  We should probably reword the Sail manual, @Alasdair?

PR #1106 (2025-07-02): Rename `ast` to `instruction`
Fixes #1018.

Should go along with an accompanying update to the Sail manual, but not a prerequisite to merging this.
closed 2025-07-02 by Timmmm