riscv/sail-riscv #1041: Dynamic detection of used extensions from ELF and generation of coverage report

nadime15, 2025-06-13:
This is based on the discussion in #870 and the idea of analyzing an ELF file to determine whether it’s a 32-bit or 64-bit binary, as well as identifying all the extensions it uses.

I think it would be cool to add a feature that does exactly that. As @jordancarlin suggested, we could support two modes:

1)  One where you provide a config file that exactly matches the DUT’s specification,

2)  And one where the tool tries to figure out the used extensions dynamically based on the ELF and does not a config file.

As far as I can tell, there is no reliable way to figure out which extensions were actually used at compile time, except by going through the entire ELF, parsing every instruction, and even going deeper into CSR usage (e.g., writes to satp, etc.). And even then, we might miss extensions that were part of the -march string but never used (those extensions should not matter in this case?). 

This also raises the question. Should we include extensions that are implicitly enabled (like Sv32, Sv39, etc.) or just stick to the ones that are explicitly part of -march?

While doing that analysis, we could also generate a coverage report, similar to what this [project](https://github.com/davidcastells/riscvisacoverage) does:

https://davidcastells.github.io/riscvisacoverage/test_vmlinux.html

Which I think is the real interesting part. It shows how many instructions were found, the number of occurrences of a given instruction, which ELF sections exists, what extensions were used, and so on, that kind of insight could be useful for benchmarking or code optimization.

To parse the ELF itself, we could use something like [https://github.com/serge1/ELFIO/tree/main](https://github.com/serge1/ELFIO/tree/main), a C++ header-only library for reading and generating ELF files. That would let us find executable regions, at least. It will not catch runtime changes, like regions marked non-executable that become executable later but I think that is the exception rather than the rule.

I think a feature like this could be useful not only for people who want to optimize and benchmark their code, but also for those working on toolchains, simulators, or simply trying to understand how their code maps to ISA features.

Does this make sense and would there be interest in this? If so, I would start working on that.

jordancarlin, 2025-06-13:
Using `.riscv.attributes` seems like a good solution and is definitely preferable to parsing the entire ELF. Having the two modes seems useful: auto detection when it is being used as a reference model to test software and config files when it is being used as a reference model to test a DUT.

Timmmm, 2025-06-13:
Don't we only need to do this for XLEN anyway? I think almost all of the other extensions are additive and we enable them all by default.

Timmmm, 2025-06-13:
For running the ELF at least. I guess if you want automatically-configured coverage you could use the ISA string in the ELF. That sounds like it should go in the coverage tool though.

Timmmm, 2025-06-13:
What would you use the information from scanning the ELF for though? You can really use it to configure a coverage tool because then it would silently squash cases where you don't test an extension at all (e.g. if you accidentally compile for softfloat).

nadime15, 2025-06-13:
Another point: years ago I was working on a custom RISC-V extension, and part of the evaluation was just to look at the ELF before and after, see how it changed in terms of used instructions, instruction frequencies, etc.

If someone wants to implement new (custom) extensions in Sail, they could easily check their impact by running both ELFs and getting a feel for whether it actually made a difference. I think this could be helpful for research too

Timmmm, 2025-06-13:
Suppose you are trying to generate a coverage report for a particular chip. You want to cover 100% of the instructions your chip supports. Your chip supports the F extension.

You write your auto-coverage tool to figure out which extensions are in the ELF and enable bins for those extensions based on that. It reports 100% coverage! Yeay! You build your chip.

Later you find a bug in F and it turns out you never tested those instructions because you accidentally compiled for softfloat, no F instructions were in the ELF so the auto-coverage tool didn't add any bins for it.

In that example you linked, there are no F instructions, so instead of saying "Implemented instructions: 0 (0.00\%)" that section would just disappear.

So really if you're doing coverage properly you need to define what you expect to hit in some other way, and then there's no need to scan the ELFs.

jrtc27, 2025-06-13:
XLEN is easy, you just look for ELFCLASS32 vs ELFCLASS64.

.riscv.attributes is for knowing which *conflicting* extensions the binary is using (F vs Zfinx, C/Zcd vs Zcm[pt]). It's also not exactly new. But yes, maybe it's stripped (or discarded by a linker script), and maybe it doesn't accurately mirror the actual extensions used if some form of dynamic dispatch (whether IFUNCs or hand-rolled) is in use.

nadime15, 2025-06-13:
> Suppose you are trying to generate a coverage report for a particular chip. You want to cover 100% of the instructions your chip supports. Your chip supports the F extension.

I think this is simply a different use case, one I did not have in mind when describing the feature. I would not rely on this report to determine which extensions my chip should support and use it as a basis for a tape out.  

> In that example you linked, there are no F instructions, so instead of saying "Implemented instructions: 0 (0.00%)" that section would just disappear.

I have not looked deeper into the repo yet but I am not sure on what basis he added the F table without any instructions, maybe using .riscv.attributes.

The point I tried to make is basically to perform an ELF only analyses, it is not tied to any particular chips etc. The feature basically just reads the ELF and collects information.