riscv/sail-riscv #1050: Delete `riscv_hints.sail`

Timmmm, 2025-06-13:
This file causes some issues because it generates invalid assembly - #1013 and #112.

I also don't see that it is really needed. The instructions are hints but that has no semantic meaning until a meaning is assigned to it.

I think we should just delete that file. When deleting it we also need to ensure:

1. That the fence instructions correctly execute reserved fences as nops.
2. That the `encdec` for the non-hint instructions which filters out hints is removed. E.g. currently we have

```
mapping clause encdec_compressed = C_SRAI_HINT(rsd)
  <-> 0b100 @ 0b0 @ 0b01 @ encdec_creg(rsd) @ 0b00000 @ 0b01

mapping clause encdec_compressed = C_SRAI(nzui5 @ nzui40, rsd)
  <-> 0b100 @ nzui5 : bits(1) @ 0b01 @ encdec_creg(rsd) @ nzui40 : bits(5) @ 0b01
  when nzui5 @ nzui40 != 0b000000 & (xlen == 64 | nzui5 == 0b0) & currentlyEnabled(Ext_Zca)
```

This should be changed to just

```
mapping clause encdec_compressed = C_SRAI(ui5 @ ui40, rsd)
  <-> 0b100 @ ui5 : bits(1) @ 0b01 @ encdec_creg(rsd) @ ui40 : bits(5) @ 0b01
  when (xlen == 64 | ui5 == 0b0) & currentlyEnabled(Ext_Zca)
```

bacam, 2025-06-16:
This would lose both the existence of the hint instructions and that they explicitly have no visible architectural effects, both of which are pretty clear in the prose spec.  If you really want the Sail model to capture the full intended meaning of these instructions, you could add an external `hint` function to document that something outside the architectural state might happen, similar to the semantics of fence, which calls `sail_barrier` to interact with the memory model.

There's also an issue when you add CHERI capabilities.  Some of the non-hint instructions are no longer no-ops, but instead strip the capability metadata from a register.  I noticed this in practice with the cheriot-ibex design: https://github.com/microsoft/cheriot-ibex/issues/83

Timmmm, 2025-06-16:
> This would lose both the existence of the hint instructions and that they explicitly have no visible architectural effects, both of which are pretty clear in the prose spec.

But their non-hint definitions force them to have no visible architectural effects anyway (except incrementing `minstret`). Stating explicitly that hints have no visible architectural effect is redundant.

> There's also an issue when you add CHERI capabilities. Some of the non-hint instructions are no longer no-ops, but instead strip the capability metadata from a register.

Interesting. I feel like the solution here is for CHERI to say that those instructions are *not* hints for CHERI, and again to delete `riscv_hints.sail`. Or it could special-case 0, but that seems worse.

Isn't the whole point of these hint instructions that they deliberately chose instructions that *already happened* to have no architectural effect. If CHERI adds some architectural effect to them then that assumption is violated and they can't be hints any more. I'll ask around about this anyway...

PeterRugg, 2025-06-16:
> you could add an external hint function

This sounds quite tidy. Having `hint` defined as a Sail empty function and calling it in the semantic function of the compressed decoding would solve this well.

Timmmm, 2025-06-16:
I'll make an issue in the CHERI spec.

jrtc27, 2025-06-16:
But the whole point is that they are hints because they happen to be NOPs. Your implementation isn't supposed to change because they're hints unless you assign that specific NOP a special meaning.