riscv/sail-riscv #1070: Vector instructions don't check vector group alignment

Timmmm, 2025-06-19:
The instructions apparently don't check that e.g. for LMUL=4 you don't use `vs1=x3` - you have to use a multiple of LMUL for the register index.

We need to add a function like this (not tested):

```
function misaligned_vreg(Vregidx(vreg) : vregidx, EMUL_pow : range(...)) -> bool =
  EMUL_pow > 0 & vreg[EMUL_pow - 1 .. 0] != zeros()
```

And then add it all over the place.

Relevant spec:

> When LMUL=2, the vector register group contains vector register v n and vector register v n+1, providing twice the vector length in bits. Instructions specifying an LMUL=2 vector register group with an odd-numbered vector register are reserved.
>
> When LMUL=4, the vector register group contains four vector registers, and instructions specifying an LMUL=4 vector register group using vector register numbers that are not multiples of four are reserved.
>
> When LMUL=8, the vector register group contains eight vector registers, and instructions specifying an LMUL=8 vector register group using register numbers that are not multiples of eight are reserved.

nadime15, 2025-07-15:
I took a look at this. Do people prefer to add this type of function (misaligned_vreg()) as part of the encdec clause, or should it be part of the execute clause? Either way, this will bloat the code.

Assuming we add this to the execute clause, would it make sense to define a more general function that checks all possible things at once? We already have 1-3 checks for tons of vector instructions (`illegal_variable_width()`, `valid_reg_overlap()`, `init_masked_result()`, etc.) and now possibly `misaligned_vreg()` as well which all can trigger a trap.

I think it makes sense to define a function like `validate_vector_instruction()` that in turn calls all the other functions (similar to what @pmundkur @Alasdair did for the generalization of the vmem interface).

This would also fit in with @Timmmm's email from today, which is about adding Sail code into the specification and cleaning it up/shortening it a little.

Any thoughts?

Timmmm, 2025-07-15:
Yeah I think anything that is true for *all* vector instructions (or at least the majority) should really go in functions if possible. 

Not sure about decoder vs execute. I guess it depends how simple you can make it - we probably don't want super complicated `when` expressions.