riscv/sail-riscv #1085: Vector Load/Store Whole Register Instructions are missing

ThinkOpenly, 2025-06-23:
The following instructions are currently not present:
- vl{lmul=1,2,4,8}re{elen=8,16,32,64}.v
- vs[lmul=1,2,4,8}r.v

(Or at least I couldn't find them.)

jordancarlin, 2025-06-23:
Pretty sure they're here: https://github.com/riscv/sail-riscv/blob/d7847673f73dcaf91a72fa3f8a33d6771a433e33/model/riscv_insts_vext_mem.sail#L571C1-L700C101?

ThinkOpenly, 2025-06-23:
They're not.

```
"vl" ^ nfields_string(nf) ^ "re" ^ vlewidth_bitsnumberstr(width) ^ ".v" ^ spc() ^ vreg_name(vd) ^ sep() ^ "(" ^ reg_name(rs1) ^ ")"
```
`nfields_string` is either the empty string or `seg`something, but it's not an "lmul" value {1,2,4,8}.

jordancarlin, 2025-06-23:
Looks like those are the correct instructions, they just have the wrong assembly. Fixed in #1088.

commit 304b8c52 (2025-06-24): Fix vector load/store whole registers instructions assembly (#1088)

The whole register vector load/store instructions should use `lmul` in
the assembly mnemonic, not `seg...`.

Fixes #1085.
PR #1088 (2025-06-24): Fix vector load/store whole registers instructions assembly
The whole register vector load/store instructions should use `lmul` in the assembly mnemonic, not `seg...`.

Fixes #1085.
closed 2025-06-24 by Timmmm