riscv/sail-riscv #1111: The ACT test failed when the pmp count was 32

penghuiho, 2025-06-30:
Could it be that the "sail" function does not support "--pmp-count=32"? Here is the test command：
riscv_sim_RV64 --pmp-count=32
PMP count: 32
invalid PMP count: must be 0, 16 or 64

File1 Path:/riscof_work/pmp64/src/pmp64-CFG-reg.S/dut/DUT-Fpga.signature
File2 Path:/riscof_work/pmp64/src/pmp64-CFG-reg.S/ref/Reference-sail_c_simulator.signature
Match  Line#    File1    File2
*          2 9f9f9f9f deadbeef
*          3 9f9f9f9f deadbeef
*          4 9f9f9f9f deadbeef
*          5 9f9f9f9f deadbeef
*          6 000001ff 00000000
           7 00000000 00000000
*          8 000001ff 00000000
           9 00000000 00000000

3.7.1. Physical Memory Protection CSRs
PMP entries are described by an 8-bit configuration register and one MXLEN-bit address register. Some PMP settings additionally use the address register associated with the preceding PMP entry. Up to 64 PMP entries are supported. Implementations may implement zero, 16, or 64 PMP entries; the lowest-numbered PMP entries must be implemented first. All PMP CSR fields are WARL and may be read-only zero. PMP CSRs are only accessible to M-mode

Timmmm, 2025-06-30:
Yes, as stated in the spec that you quoted, the RISC-V spec requires that you have exactly 0, 16, or 64 PMPs. It isn't valid to implement 32 PMPs. What you *can* do is implement 64 PMPs and then make half of them read-only zero. We don't support that yet though. Do the ACTs test that configuration?

penghuiho, 2025-07-01:
> Yes, as stated in the spec that you quoted, the RISC-V spec requires that you have exactly 0, 16, or 64 PMPs. It isn't valid to implement 32 PMPs. What you _can_ do is implement 64 PMPs and then make half of them read-only zero. We don't support that yet though. Do the ACTs test that configuration?

The configuration is 32, and the sail report shows the above error, but spike does not have this limitation

allenjbaum, 2025-07-01:
> I checked the Spike code - it implements this:

Until the runtime configuration machinery is set up, this (and many other
things) can't be tested by Sail.

Timmmm, 2025-07-01:
It is set up now. We just need to add the config options. I'll turn this issue into a request for this option.