riscv/sail-riscv #1125: Don't require write_CSR to return the new value

Timmmm, 2025-07-03:
Currently `write_CSR()` returns the new value of written CSRs, purely for logging. This is pretty tedious though.

Since none of the CSRs actually have read side effects, instead we can just call `read_CSR()` again for the logging.

I think the one exception might be the `seed` CSR which is ... weird. It might require special handling.

radimkrcmar, 2025-07-04:
The `seed` CSR explicitly forbids read-only accesses from software, so internally using a read-only accessor after the write should be fine for all CSRs.  (The update of `seed` should only happen as a side-effect of its write.)

radimkrcmar, 2025-07-04:
Yeah, I was proposing something like this, where `write_CSR` reports the next seed, and could be refactored to drop the last line if we used `read_CSR` in `do_CSR` instead:
```
diff --git a/model/riscv_zkr_control.sail b/model/riscv_zkr_control.sail
index e672173ed09e..9cb7e3c8f360 100644
--- a/model/riscv_zkr_control.sail
+++ b/model/riscv_zkr_control.sail
@@ -25,6 +25,8 @@ mapping opst_code : seed_opst <-> bits(2) = {
   DEAD <-> 0b11
 }
 
+register seed : bits(16) // TODO: initialize
+
 /* Entropy source spec requires an Illegal opcode exception be raised if the
  * seed register is read without also being written. This function is only
  * called once we know the CSR is being written, and all other access control
@@ -33,12 +35,14 @@ mapping opst_code : seed_opst <-> bits(2) = {
 function read_seed_csr() -> xlenbits = {
   let reserved_bits : bits(6) = 0b000000;
   let custom_bits : bits(8) = 0x00;
-  let seed : bits(16) = get_16_random_bits();
   zero_extend(opst_code(ES16) @ reserved_bits @ custom_bits @ seed)
 }
 
 /* Writes to the seed CSR are ignored */
-function write_seed_csr () -> xlenbits = zeros()
+function write_seed_csr () -> xlenbits = {
+  seed = get_16_random_bits();
+  read_seed_csr()
+}
 
 /* CSR mapping */
 mapping clause csr_name_map = 0x015  <-> "seed"
```

Timmmm, 2025-07-04:
Oh yeah I like that.