riscv/sail-riscv #1133: Test against riscv-arch-test

Timmmm, 2025-07-09:
We should support testing against [riscv-arch-test](https://github.com/riscv-non-isa/riscv-arch-test) (AKA ACTs), and run these tests in CI.

This was previously discounted because they test against Sail, but that only happens after PRs have been merged and we would like to detect issues before that (see e.g. #1132).

Another benefit would be for testing new extension, e.g. see [here](https://github.com/riscv/sail-riscv/pull/285#issuecomment-3013232514) - I would like to be able to easily run the Zacas tests in riscv-arch-test.

Unfortunately setting up riscv-arch-test involves a lot of very messy Python and YAML nonsense, so it is a lot of work.

We should have some kind of script that users and CI can run that does that hard work for us.

jordancarlin, 2025-07-09:
There is ongoing work towards a significant refactor of riscv-arch-test that (amongst other things) should make it much easier to run and setup. We're hoping to have the first version of that upstreamed within the next few months. It might make sense to wait for that.

jordancarlin, 2025-07-11:
> > There is ongoing work towards a significant refactor of riscv-arch-test that (amongst other things) should make it much easier to run and setup. We're hoping to have the first version of that upstreamed within the next few months. It might make sense to wait for that.
> 
> Is there public information about this for others to review? I've given feedback in the past about the (very reminiscent of hardware "frameworks" like AWS F1) copy-and-paste design for riscv-arch-test, and its historic ties to a GNU toolchain rather than accommodating LLVM. (And I'm at least partly responsible for blocking past attempts to merge the mess needed to support various incarnations of the ACTs in this repo...)

The current working version is available at https://github.com/openhwgroup/cvw-arch-verif. There are a number of changes still ongoing, and that repo is currently halfway between the new and old approaches, so it is not yet indicative of the planned final state of things. There is also significant work going on to make the whole flow easier to run, but that is not yet upstreamed to the repo. 

There should be minimal "copy-and-paste" like design with most of the coverpoints/tests generated from CSV files and templates instead. Compatibility with LLVM is also a priority, and it should be easy to set the desired compiler in a single spot using standard `CC` type variables. If you have specific concerns/questions/feedback, please reach out, and we're happy to discuss in more detail. A lot of conversation surrounding this is also taking place in the ACT and CSC meetings.

Timmmm, 2025-07-12:
Is that repo intended to replace `riscv-arch-test` then? It seems focused on Wally?

Timmmm, 2025-07-13:
Ah interesting. Is there a plan for how it's going to work? If we can replace `riscv-arch-test` with something better that would be great, but it's going to affect the whole community so it would be nice to ensure it meets everyone's requirements - `riscv-arch-test` had numerous issues that meant it didn't:

* Insistence on driving everything itself (we worked around this by having a fake target that just wrote the ELF filenames to a file.
* It's a total mess.
* It's quite complex to set up.
* It's not reproducible. Lots of the tests are generated but it's impossible to regenerate them - instructions are missing, and PRNGs weren't deterministically seeded.
* Generating those tests is extraordinarily slow (yeay Python).
* The issues @jrtc27 mentioned.

I'm sure you've thought of all that because you're smart and have good taste! It would be reassuring if the plan was written down somewhere though.

allenjbaum, 2025-07-14:
I completely don't understand the regeneration comment. 
Are you trying to regenerate the .S files, or the binaries from the .S files?

Timmmm, 2025-07-14:
Regenerate the .S files. The seeding issue would be fixed by [this](https://github.com/riscv-non-isa/riscv-arch-test/pull/608) but that can't be tested because of a bug in riscv-isac - clearly it has not even been run for a long time. The lack of instructions ... was sort of solved by someone roughly explaining it in a random Github issue comment a while back (that I'll probably never be able to find again!). But it relied on files like `riscv-ctg/sample_cgfs/dataset.cgf` (why is this a "sample"?) and I think those files have actually been deleted now.

It would be really good to have less of a mess and use modern software best practices for any replacement:

* Run everything in CI.
* Everything should be reproducible.
* If using Python 🤮 at least enable linters, type checkers, etc. and use [a modern Python package structure](https://github.com/riscv/riscv-opcodes/pull/294#issuecomment-2394991768).

@jordancarlin is young and clever so I'm sure he's thought of all this, but as I said - it would be nice to see it written down so we can be sure!

jrtc27, 2025-07-14:
Even more convenient is not using third party libraries at all. For this kind of thing I don’t see why you’d need one?