riscv/sail-riscv #1174: config_print_instr is checked twice

Timmmm, 2025-07-29:
This flow is weird...

```
        if   get_config_print_instr()
        then {
          print_instr("[" ^ dec_str(step_no) ^ "] [" ^ to_str(cur_privilege) ^ "]: " ^ BitStr(PC) ^ " (" ^ BitStr(w) ^ ") " ^ to_str(instruction));
        };

...

unit print_instr(sail_string s)
{
  if (config_print_instr)
    fprintf(trace_log, "%s\n", s);
  return UNIT;
}
```

We check `config_print_instr` twice. I would say since we are checking it in Sail we don't need to check it again in C. And it makes sense to check it in Sail because we avoid the overhead of string formatting if it's disabled. So we should probably remove the C checks.

Probably the same issue for other `config_print..`s but I haven't checked.

arichardson, 2025-07-29:
I think there is no need to have `print_instr() and print_platform()` anymore, should probably just have `trace()/print_trace()` and make sure all calls are guarded.

pmundkur, 2025-08-11:
`config_trace_xxx`  or `trace()/print_trace()` might make adding the Trace extension confusing.
https://github.com/riscv-non-isa/riscv-trace-spec/releases
Is there a better name to talk about the current execution logging when we update the docs?

Timmmm, 2025-08-11:
Maybe "log"?

commit 28c80a32 (2025-08-12): Remove duplicate config_print checks (#1197)

The config flags `config_print_instr` and `config_print_platform` are
checked twice in the Sail and C code. This commit removes the C
functions `print_instr()` and `print_platform()`, and replaces them with
a single function `print_trace()`, which doesn't do a redundant check.

Fixes #1174
PR #1197 (2025-08-12): Remove duplicate config_print checks
The config flags `config_print_instr` and `config_print_platform` are checked twice in the Sail and C code. This commit removes the C functions `print_instr()` and `print_platform()`, and replaces them with a single function `print_trace()`, which doesn't do a redundant check.

Fixes #1174
closed 2025-08-12 by Timmmm