riscv/sail-riscv #1180: Move first party tests to sail-riscv-tests repo

jordancarlin, 2025-07-31:
Now that we have a repo dedicated to compiling tests for the Sail model, we should migrate the first-party tests to it and precompile them like we do for other third-party tests. This avoids the need for an installed cross-compiler while developing the model.

Timmmm, 2025-07-31:
But then we lose the ability to easily add these tests together with the corresponding fixes. You'd have to now write a new test, add it to the tests repo, wait for a new release, update the Sail model to point at it, and *finally* make your bug fix PR.

Timmmm, 2025-07-31:
Plus it's quite nice to have a place in the Sail model where the compiler and environment is set up so you can easily write some C/assembly and run it on the model.

closed 2025-08-08 by jordancarlin