riscv/sail-riscv #1239: Zalrsc (sc.w/d) checking reservation before checking pmp permission.

sudo-apt-abdullah, 2025-08-29:
A region with no permission (XWR-000) from 0x80002000 to 0x80002040 is there and we're trying to lr.w from start of the region.

```
[233] [M]: 0x800004A0 (0x1007A62F) lr.w x12, (x15)
trapping from M to M to handle load-access-fault
handling exc#0x05 at priv M with tval 0x80002000
CSR mstatus (0x300) <- 0x00001800
CSR mstatush (0x310) <- 0x00000000
CSR mcause (0x342) <- 0x00000005
CSR mtval (0x343) <- 0x80002000
CSR mepc (0x341) <- 0x800004A0 
```
Since, there's no permission read permission, it throws access-fault and when we're trying to sc.w at the same address, with no write permission it's checking reservation first and since no load occur at that address, it simply putting 1 in rd. But it should check the pmp permission first and throw access-fault.

```
[345] [M]: 0x800004AC (0x18C7A62F) sc.w x12, x12, (x15)
x12 <- 0x00000001
```

KotorinMinami, 2025-08-31:
To my knowledge, the `lr.w` instruction, upon successful execution, registers a reservation set. When `sc.w` is used, it checks if the target has a corresponding reservation set. If no such set exists, it will not access memory. In the scenario you described, `sc.w` will not access memory, and therefore will not throw an access-fault.

KotorinMinami, 2025-08-31:
This is my mistake. I must have overlooked the spec.

davidharrishmc, 2025-10-17:
@Timmmm are you the one who would upstream this fix?  Is there a timeframe we should expect?

closed 2026-01-30 by Timmmm

jordancarlin, 2026-02-02:
Confirmed with the test authors that #1500 fixed this.