riscv/sail-riscv #1250: Add syscall support

trdthg, 2025-09-02:
I've been working on this recently，the target is to making sail-riscv able to run user programs like qemu-user/spike/[NEMU](https://github.com/OpenXiangShan/NEMU)

Currently I'm testing riscv-pk with spike (thanks to #1102)

related issues

- #252
- #346 

I have some code locally and will push incrementally.

trdthg, 2025-09-02:
I think this doesn't go against the model.
Implementing syscall will only insert one dispatch_syscall function on the sail side, and the rest are C stuff, which are separated from the model. Also this won't require too much code.

jrtc27, 2025-09-02:
But what syscalls? Are you implementing some toy OS set of syscalls? Are you reimplementing some existing OS's syscalls? Or are you translating to the native OS's syscalls (and if so, which OSes do you support)? None of those are at all a part of the architecture. They are all part of a software stack that sits on top of the architecture that you are choosing to bake into the model, and that applies even if written in C. This is not an implement-an-OS repository, this is an implement-an-architecture repository.

Timmmm, 2025-09-02:
Yeah I'm not sure about this either. HTIF isn't exactly well specified. Also I don't really understand why you would want to run another emulator *inside* the Sail emulator?

I do think we want proper terminal IO but I'd much rather that was done by implementing a *real* UART device - probably AXI UART Lite. That is very simple but it also requires PLIC support... which is also fairly simple but a bit more work so I haven't got around to it yet.

What syscalls would you want apart from terminal IO? Do you want host filesystem access?

arichardson, 2025-09-02:
The HTIF syscall stuff is not particularly well documented, if you want to run some binaries that rely on semihosting, it would IMO be better to support the RISC-V semihosting spec (https://github.com/riscv-non-isa/riscv-semihosting) which would allow running e.g. test binaries built with [picolibc](https://github.com/picolibc/picolibc)'s -lsemihost.

arichardson, 2025-09-03:
If you are planning to implement semihosting here is the GEM5 implementation for reference: https://github.com/gem5/gem5/blob/stable/src/arch/generic/semihosting.cc and QEMU is here https://github.com/qemu/qemu/blob/master/semihosting/syscalls.c